Jasonchenlijian/FastBle #341: oppo5.0 5.1的某些机子能写能读，就是收不到通知

bossAnmi, 2018-11-02:
setCharacteristicNotification 里返回了true。求解

bossAnmi, 2018-11-02:
追加：连接前必须配对吗。我这不配对，连接30是后就自动断开了，求解

Jasonchenlijian, 2018-11-05:
setCharacteristicNotification 返回true表示设置成功了。
连接前不需要配对，30秒断开是某些设备一段时间不进行通信会自动断开。

bossAnmi, 2018-11-05:
如标题，setCharacteristicNotification返回true了，可上述某些机型onCharacteristicChanged不回调

yuwoquan, 2018-11-16:
这个你解决了吗。我目前也是，无法收到消息

Github-header, 2020-09-24:
> 如标题，setCharacteristicNotification返回true了，可上述某些机型onCharacteristicChanged不回调

请问你之后怎么解决的

bossAnmi, 2020-09-24:
> > 如标题，setCharacteristicNotification返回true了，可上述某些机型onCharacteristicChanged不回调
> 
> 请问你之后怎么解决的

某些低版本机型用经典蓝牙开发代替ble，可以收到通知

Github-header, 2020-09-24:
> > > 如标题，setCharacteristicNotification返回true了，可上述某些机型onCharacteristicChanged不回调
> > 
> > 
> > 请问你之后怎么解决的
> 
> 某些低版本机型用经典蓝牙开发代替ble，可以收到通知

好的，谢谢了