Jasonchenlijian/FastBle #348: 写入失败GattException{gattStatus=133} BleException { code=101, description='Gatt Exception Occurred! '}

lijore, 2018-11-14:
这个什么问题

Jasonchenlijian, 2018-11-28:
我觉得你可能需要在两次操作之间延时一小段时间，也许会解决你的问题。

fangood, 2019-01-14:
133应该是适配问题，
`BluetoothDevice bluetoothDevice= mDatas.get(position);
                    //连接设备
                    tvSerBindStatus.setText("连接中");
                    if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.M) {
                        mBluetoothGatt = bluetoothDevice.connectGatt(MainActivity.this,
                                true, gattCallback, TRANSPORT_LE);
                    } else {
                        mBluetoothGatt = bluetoothDevice.connectGatt(MainActivity.this,
                                true, gattCallback);
                    }`

523931340, 2020-10-05:
请问这个问题解决了么