Jasonchenlijian/FastBle #353: 获取蓝牙名称的两种方法的疑惑

SJL-Code, 2018-11-23:
Android 官方提供了两种方法获取蓝牙的名称，如下
// Returns the local name of the BLE device. The is a UTF-8 encoded string
result.getScanRecord().getDeviceName()
// Get the friendly Bluetooth name of the remote device
result.getDevice().getName()

而大佬你提供的方法是 public void onScanning(BleDevice bleDevice)  bleDevice.getName() 获取蓝牙名称的, 这个对应Android 官方的貌似是result.getDevice().getName()，因为我们硬件那里广播出来的蓝牙名称是会动态改变的，但是提供这种方法获取的名称貌似不会动态改变，所以想问有没有提供对应官方result.getScanRecord().getDeviceName() 获取蓝牙名称的方法，因为这样获取的名称会动态改变，谢谢。

Jasonchenlijian, 2018-12-31:
当你调用scan方法进行扫描的时候，BleScanCallback中的onLeScan，会不断刷新扫描到的周围外设的信息，通过BleDevice的getScanRecord，可以得到广播数据byte[]，这里面包含了你所说的动态变化的外设名称。只有如何解析，目前版本SDK中并没有提供，可以自行解析一下。

SJL-Code, 2019-01-09:
@Jasonchenlijian 谢谢了，已经在android蓝牙源代码中找到了解析广播数据byte[]的方法了。

renwfy, 2019-06-19:
@SJL-Code ，具体怎么实现的，代码可以分享下吗

SJL-Code, 2019-06-24:
> @SJL-Code ，具体怎么实现的，代码可以分享下吗

private static final int DATA_TYPE_LOCAL_NAME_COMPLETE = 0x09;
    private static final int DATA_TYPE_SERVICE_UUID_16_BIT_PARTIAL = 0x02;

    /**
     * 通过解析"BLE扫描的记录"获取蓝牙的本地名称
     *
     * @param scanRecord BLE扫描的记录
     * @return 蓝牙的本地名称
     */
    public static String parseScanRecord(byte[] scanRecord) {
        int currentPos = 0;
        String localName = null;
        try {
            while (currentPos < scanRecord.length) {
                int length = scanRecord[currentPos++] & 0xFF;
                if (length == 0) {
                    break;
                }
                int dataLength = length - 1;
                int fieldType = scanRecord[currentPos++] & 0xFF;
                if (DATA_TYPE_LOCAL_NAME_COMPLETE == fieldType || DATA_TYPE_SERVICE_UUID_16_BIT_PARTIAL == fieldType) {
                    localName = new String(extractBytes(scanRecord, currentPos, dataLength));
                }
                currentPos += dataLength;
            }
        } catch (Exception e) {
            LogUtil.e(TAG, "无法解析的扫描记录: " + Arrays.toString(scanRecord));
        }
        return localName;
    }

    private static byte[] extractBytes(byte[] scanRecord, int start, int length) {
        byte[] bytes = new byte[length];
        System.arraycopy(scanRecord, start, bytes, 0, length);
        return bytes;
    }

WhiteCjy, 2019-09-04:
用这个获取到名称的还是为Null，华为9.0