Jasonchenlijian/FastBle #361: 收不到notify，是什么情况呢？

lanmanck, 2018-12-11:
我的设备一个特征，可以read,write_no_resp,notify,现在write蓝牙芯片收到了，但是硬件返回来的数据收不到，notify也没有响应，是什么情况呢？
设备没有问题的，还有一套程序用别人的可以正确读写。

Jasonchenlijian, 2018-12-11:
有没有先打开notify，然后再write？

lanmanck, 2018-12-11:
有打开notify，在ondeviceconnect时就延时100ms：
`
 @Override
    protected void onDeviceConnected() {
        super.onDeviceConnected();

        ViseLog.e("连接成功");

        Toast.makeText(mContext,mContext.getString(R.string.ble_connect_ok),Toast.LENGTH_LONG).show();
        mAdapter.notifyDataSetChanged();
        progressVisibility.set(View.GONE);
        enableBtn.set(true);

        // 设置BLE监听机制？回调在 bluetoothrequestlistener.java
        handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                ViseLog.i("setUpNotify");
                mModel.setUpNotify();
            }
        }, 100);

        handler.postDelayed(new Runnable() {
            @Override
            public void run() {
                ViseLog.i("开始获取序列号");
                sendData(TaluerUtils.getSendData(
                        0x1B,
                        TaluerUtils.APP_BLE_CH, TaluerUtils.REMOTE_FRAME, TaluerUtils.encrypt(null)));
                handler.postDelayed(retrySendData, 500);
            }
        }, 200);
    }
`
其中mModel.setUpNotify();的代码是：
`BleManager.getInstance().notify(
                mBleDevice,
                Constants.UUID_SERVICE,
                Constants.UUID_NOTIFY,
                BluetoothRequestListener.getInstance());`
写的代码是：
`BleManager.getInstance().write(
                    mBleDevice,
                    Constants.UUID_SERVICE,
                    Constants.UUID_NOTIFY,
                    data,
                    new BleWriteCallback() {
                        @Override
                        public void onWriteSuccess(int current, int total, byte[] justWrite) {
                            ViseLog.d("正在发送：" + HexUtil.formatHexString(justWrite));
                            ViseLog.d("发送成功，current: "+ current +", total: " + total);
                        }

                        @Override
                        public void onWriteFailure(BleException exception) {
                            ViseLog.e("发送失败");
                        }
                    });`

下面可以收到数据，就是这3个回调：
// 这些回调来自setUpNotify()
   ' @Override
    public void onNotifySuccess() {
        ViseLog.i("监听回复成功");
    }

    @Override
    public void onNotifyFailure(BleException exception) {
        ViseLog.i("监听回复失败");
    }

    @Override
    public void onCharacteristicChanged(byte[] data) {
        ViseLog.i("onCharacteristicChanged()");
        ViseLog.d(HexUtil.formatHexString(data));
        for (byte aData : data) {
            processInputData(aData);
        }
    }'
只有 onNotifySuccess有反应，其他2个没反应了。

lanmanck, 2018-12-11:
换了个安卓8.0的手机，又可以了，之前是安卓4.4，华为荣耀4，绝对是哪里的兼容性问题呢？

lanmanck, 2018-12-12:
安卓7.0的华为荣耀7也不行，无语呀。。。

closed 2021-12-07 by Jasonchenlijian