Jasonchenlijian/FastBle #393: 数组越界的空指针异常 BleScanner.java:84

AndyGu, 2019-03-25:
java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
        at java.util.ArrayList.get(ArrayList.java:437)
        at com.clj.fastble.d.c$1$1.run(BleScanner.java:84)
        at android.os.Handler.handleCallback(Handler.java:873)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:214)
        at android.app.ActivityThread.main(ActivityThread.java:7032)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:493)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:964)

源码中的BleManager.getInstance().connect(list.get(0), callback); 其中list应做为空检查，跪求po主改下

KomoriWu, 2019-03-27:
+1 刚想提这个，bugly报了几十次这个了

JinZhou198877, 2019-04-28:
+1，友盟上报了几次，机型OPPO R11 Pluskt 7.1.1和ATU-AL10 8.0.0