Jasonchenlijian/FastBle #394: 读特征下面的描述符

KomoriWu, 2019-03-29:
框架提示:this characteristic not support read!

KomoriWu, 2019-03-29:
框架不支持，只能修改源码了 
  List<BluetoothGattService> services = BleManager.getInstance().getBluetoothGattServices(device);
                        BluetoothGattCharacteristic characteristic = BleManager.getInstance().getBluetoothGattCharacteristics(services.get(3)).get(0);

                        // //00002901-0000-1000-8000-00805f9b34fb
                        BleManager.getInstance().getBluetoothGatt(device).readDescriptor(characteristic.getDescriptors().get(0));

然后重写 BleBluetooth.java 里面BluetoothGattCallback
 onDescriptorRead

closed 2019-03-29 by KomoriWu

xiaojidonggong, 2019-12-09:
没太懂，到底上面代码代表了 什么意思。
我和你遇到了相同的错误