Jasonchenlijian/FastBle #404: 连接 加密的BLE 设备时，不配对也可以进行连接和读写数据

githubforzjm, 2019-04-28:
你好
 感谢你的FastBle这个开源库，使用起来非常方便，
但在使用FastBle时发现连接到 加密的BLE设备时，系统会弹出配对请求，但是在忽略配对请求的情况下，照样能够正常建立连接和读写数据，  我也是使用了Google 的[API ](https://developer.android.com/guide/topics/connectivity/bluetooth-le#java) 也存在这种情况，

如果是这样，BLE设备加密还有什么意义呢？  同样发现IOS建立BLE连接时，如果不允许配对请求的话，就无法建立连接，一直阻塞知道允许配对请求

请问针对这种情况，有什么好的解决版本

Jasonchenlijian, 2019-05-15:
请问你的加密指的是哪方面的加密？

githubforzjm, 2019-05-15:
> 请问你的加密指的是哪方面的加密？
@Jasonchenlijian   加密指的是支持BLE的设备是加密的

[关于这个现象有人提到过](https://www.jianshu.com/p/71116665fd08)
大多数BLE设备是非加密的，也就是不需要与手机配对即可通信。大部分软件都是同步BLE设备的中数据到手机端，整个流程也是自动执行的。大体流程就是：扫描目标设备、开启连接、连接成功扫描服务，扫到服务开启监听、收发数据、断开连接。
BLE设备加密后在连接设备时会弹出系统蓝牙配对对话框，等待用户配对。按常理说此时设备是没有建立连接的，因为还没有配对成功，但是安卓手机端不仅可以完成建立连接，而且还能扫描到服务。如果按照自动流程就去开启监听是无法成功的，因为设备还没有配对成功。