Jasonchenlijian/FastBle #408: 关于扫描筛选指定服务的设备 取得是并集

BestCoderXQX, 2019-05-14:
BleScanRuleConfig scanRuleConfig = new BleScanRuleConfig.Builder()
                .setServiceUuids(serviceUuids)      // 只扫描指定的服务的设备，可选

这里uuid的集合 比如有3个uuid， 扫描的结果要都满足这3条
可不加一个方法 实现 我提供若干个uuid， 只要扫描出的结果满足其中的一条就可以作为结果

Jasonchenlijian, 2019-05-15:
你的建议很好，可以尝试增加。

houyuehai, 2019-05-16:
我遇到这个问题了，公司有好几种uuid，目前看来只能在结果中筛选了

houyuehai, 2019-05-16:
> BleScanRuleConfig scanRuleConfig = new BleScanRuleConfig.Builder()
> .setServiceUuids(serviceUuids) // 只扫描指定的服务的设备，可选
> 
> 这里uuid的集合 比如有3个uuid， 扫描的结果要都满足这3条
> 可不加一个方法 实现 我提供若干个uuid， 只要扫描出的结果满足其中的一条就可以作为结果

请问你后来是怎么实现这个功能的。

BestCoderXQX, 2019-05-16:
没有实现  最麻烦的就是每次扫描扫一个uuid  多扫描几次  取交集

houyuehai, 2019-05-16:
public ScanFilter.Builder setServiceUuid（ParcelUuid serviceUuid， 
                 ParcelUuid uuidMask）

在部分服务uuid​​上设置过滤器。这uuidMask是比特掩码 serviceUuid。将掩码中的任何位设置为1表示该位需要匹配serviceUuid，0表示忽略该位。
我看到了这个api。不过是5.1才有的功能。

BestCoderXQX, 2019-05-16:
https://github.com/Jasonchenlijian/FastBle/issues/274  适用该方法解决