Jasonchenlijian/FastBle #433: 有人遇到过 偶尔会出现一台设备也搜索不到的情况 这时候关闭手机蓝牙再重新打开才可以 好几台手机都偶尔会出现这种情况 这是什么问题

yydnjas, 2019-08-22:
如题

kscMaster, 2019-09-10:
出现在低端机上的问题吧，跟蓝牙也有关系，我们公司用国产银尔达的蓝牙，有些低端机就扫不出来，换了国外的一家，就没这个问题了

yydnjas, 2019-09-10:
> 出现在低端机上的问题吧，跟蓝牙也有关系，我们公司用国产银尔达的蓝牙，有些低端机就扫不出来，换了国外的一家，就没这个问题了

不是低端机  一开始都搜得到  可能是异常退出gatt没有关闭超过数量  然后导致蓝牙都搜索不到  关闭蓝牙再重新打开功能就恢复了

yydnjas, 2019-09-17:
> 我也遇到这种问题，解决了吗

改成每次扫描10秒  然后间隔3秒再扫描一次  暂时没出现这种问题  还需要多测试

yydnjas, 2019-09-23:
> 我也遇到这种问题，解决了吗

刚刚又浏览了一个帖子  有在说安卓在7.0之后BLE增加一个防滥用的功能---即不要在30s内对蓝牙扫描
重复开启-关闭超过5次。所以可以在搜索的时候用一个flag来控制扫描的线程来防止重复扫描。这也验证了我上面说的  就是将扫描时候改短  然后分为多次扫描来避免重复扫描

ruizhang1015, 2019-09-23:
现在你再也没出现过这种问题了吗？我现在还出现过设备一直连不上的问题，也要重启蓝牙，就能连上了

ruizhang1015, 2019-09-23:
而且我用的是BluetoothLeScanner扫描的，开始扫描后就不会停止，没有你说的扫描持续时间和间隔的问题

yydnjas, 2019-09-24:
> 而且我用的是BluetoothLeScanner扫描的，开始扫描后就不会停止，没有你说的扫描持续时间和间隔的问题

一定要持续扫描吗？持续扫描容易出现这种问题  现在最好的解决办法就是间断搜索  我之前出现问题也是持续扫描才出现的问题  测试的时候之前的扫描还没关掉就开始新的扫描  这样容易出问题

closed 2021-12-07 by Jasonchenlijian