Jasonchenlijian/FastBle #440: 设备断开连接后不能再被扫描到

aninc, 2019-09-05:
频繁操作蓝牙连接和断开 会导致设备扫描不到。
现象：
用其他中心设备也无法扫描到设备，然后重新关闭打开蓝牙或者杀掉自己的进程，就能扫描到。
好像是没有成功断开连接 ，但onDisconnected()回调有调用，getAllConnectedDevices无设备。
排除了线程问题 都是在主线程操作的。
不确定是lib的问题还是手机的问题 有几个测试机出现过这个现象。(虽然我觉得是手机垃圾的原因)

dilemma1212, 2019-12-09:
请问下解决了没，我这边也遇到了这个问题

kalshen2018, 2020-07-01:
我也遇到了同样的问题。

lnuvs, 2020-07-21:
> 请问下解决了没，我这边也遇到了这个问题

我记得没有解决掉 早离职了(233

lnuvs, 2020-07-21:
> 我也遇到了同样的问题。我感觉是断开连接失败了，虽然回调走了，但系统的蓝牙断开广播没走，设备还是连接中，就是getAllConnectedDevices 搜不到设备了。
> 
> 我发现同一个设备多次连接后会出现该问题，我的处理办法是每次连接前判断该设备是否已连接，已连接就不再走真实的连接逻辑，直接回调连接成功的逻辑。先通过BleManager.getInstance().isConnected(),若果已连接，直接走成功回调，绕开connect逻辑。

这个方法是不行的 lib里保存的数据清空了 isConnected()是false的。

closed 2020-07-21 by lnuvs

topul, 2022-11-19:
> 

直接走connect逻辑是没问题的，但系统底层断不开蓝牙设备就不会继续向外广播了，想从广播里拿数据就做不到了。并且很奇怪的是，用isConnected显示是连接状态，用getAllConnectedDevice又显示没有设备连接，不管调用disconnect还是disconnectAll都是成功，但isConnected就是显示连接状态，设备也确实是处于蓝牙连接状态