Jasonchenlijian/FastBle #462: 不是刷存在感，是真的想提作者说句话，此开源框架本来就是无偿的，说的不好听，作者本就没义务一一回答这些问题，有非常多的问题完全百度都能出来的，对本块技术没任何了解，奉承拿来主义问一些完全没价值的问题，还理所当然的要作者解决这解决那，除了拿个start，估计作者是真的心累

gh1225987336, 2019-11-19:


kingiis, 2019-11-21:
精武会既视感

snail-king, 2020-07-23:
你怕不是神经病吧，如果大家都不提问题的话，这个框架怎么更新，怎么变优秀？开玩乐

Liby94, 2020-07-23:
大家确实要多提一些建设性意见框架才会变得更好，但是有些智障总是不动脑子，先搜一下再提问的常识都没有。

z3880341, 2020-08-03:
说句伤人的话，我还希望这个框架不存在。。 因为我就可以使用原生Android 蓝牙进行开发了。 否则我要配合我的同事他们使用这个框架。（因为他们是框架战士，脱离了第三方框架什么都写不出来）这个框架增加我的学习成本，还把蓝牙回调BluetoothGattCallback碎片化处理。

gh1225987336, 2020-08-06:
> 你怕不是神经病吧，如果大家都不提问题的话，这个框架怎么更新，怎么变优秀？开玩乐

这位大佬，抱歉，是您没读懂意思还是我没表达清楚？我表达出了不该提出问题的意思了吗？当然大家水平不一，我也不会去评论谁谁谁的问题怎么怎么样，不懂就问也确实是应该的，作为技术人谁都是一步步爬坑爬出来的，但是问问题之前，我们不应该去自己先查一查，努努力？明明是个学高数的地方，还在这里问一些乘法口诀表的问题，您觉得这样也能使这个框架变优秀？我也没有想到在这个技术交流之处，也会出现您这种直接开喷的大佬，我说声抱歉，让您看不顺眼了。

SUPERYAO541, 2020-08-27:
> > 你怕不是神经病吧，如果大家都不提问题的话，这个框架怎么更新，怎么变优秀？开玩乐
> 
> 这位大佬，抱歉，是您没读懂意思还是我没表达清楚？我表达出了不该提出问题的意思了吗？当然大家水平不一，我也不会去评论谁谁谁的问题怎么怎么样，不懂就问也确实是应该的，作为技术人谁都是一步步爬坑爬出来的，但是问问题之前，我们不应该去自己先查一查，努努力？明明是个学高数的地方，还在这里问一些乘法口诀表的问题，您觉得这样也能使这个框架变优秀？我也没有想到在这个技术交流之处，也会出现您这种直接开喷的大佬，我说声抱歉，让您看不顺眼了。

你又沒說錯，跟他道啥歉。
一些蠢問題本來就該自己先 google 或你們那叫百度，否則人家免費送你台車，除了介紹車內設備怎麼用，還得負責教各位開車不成。

docwei2050, 2020-11-25:
> 说句伤人的话，我还希望这个框架不存在。。 因为我就可以使用原生Android 蓝牙进行开发了。 否则我要配合我的同事他们使用这个框架。（因为他们是框架战士，脱离了第三方框架什么都写不出来）这个框架增加我的学习成本，还把蓝牙回调BluetoothGattCallback碎片化处理。

这些框架本就不维护了，还是自己折腾吧

closed 2021-12-07 by Jasonchenlijian