Jasonchenlijian/FastBle #471: 数组越位异常 java.lang.IndexOutOfBoundsException

bailiangjin, 2019-12-21:
/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.mega.magicbean, PID: 7485
    java.lang.IndexOutOfBoundsException: Index: 0, Size: 0
        at java.util.ArrayList.get(ArrayList.java:437)
        at com.clj.fastble.scan.BleScanner$1$1.run(BleScanner.java:84)
        at android.os.Handler.handleCallback(Handler.java:907)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:216)
        at android.app.ActivityThread.main(ActivityThread.java:7625)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)

langmengwork, 2020-01-07:
我也遇到了这个问题

langmengwork, 2020-01-13:
我的问题是蓝牙没有打开就开始了操作，希望可以帮到你

kingiis, 2020-04-22:
打开蓝牙前
检测蓝牙状态 位置信息 权限
没有任何问题发现