Jasonchenlijian/FastBle #473: Duration of the Connection won t last longer.

VinayTShetty, 2020-01-08:
Hi thank you in frist hand for providing this library.
i was customizing this library for multiple connection.
I am having issues in maintaining the connection to the Connected Device.

Ex:-If i connect to "A" device ,after a particular interval it will disconnect.
      This is repeated for every time, when i connect to "A" device
      
     can u please help me in addressing this issue...
Thanks in advance

kingiis, 2020-04-22:
I also have this problem. According to the company's hardware engineer, it's actually a Bluetooth board with a dormant program that needs heartbeat maintenance. This is a hardware problem

kingiis, 2020-04-22:
I also won a bottle of ice coke for this bet