Jasonchenlijian/FastBle #477: No reliable write support

taimour1446, 2020-03-30:
Please include reliable write support for this library as I cannot see it anywhere

kingiis, 2020-04-03:
BleManager.getInstance().connect(bleDevice, new BleGattCallback() {
 @Override
public void onConnectSuccess(BleDevice bleDevice, BluetoothGatt gatt, int status) {

//I'm connected here. Find your data service and channel according to your Bluetooth protocol
                    String uuid = context.getInfo().cmut_uuid;//Server data
                    String uuid_w = context.getInfo().cmut_uuid_w;//Server data
                    String uuid_n = context.getInfo().cmut_uuid_r;//Server data

    App.UUID_SERVICE = gatt.getService(UUID.fromString(uuid));
                    UUID_SERVICE = App.UUID_SERVICE;

                    App.UUID_WRITE = App.UUID_SERVICE.getCharacteristic(UUID.fromString(uuid_w));
                    UUID_WRITE = App.UUID_WRITE;

                    App.UUID_NOTIFY = App.UUID_SERVICE.getCharacteristic(UUID.fromString(uuid_n));
                    UUID_NOTIFY = App.UUID_NOTIFY;
//Fire at will

taimour1446, 2020-04-03:
I want to get response data from ble when I write to ble device and get device response data on OnWriteSuccess. How this gonna help me in that situation? Can you explain further?

kingiis, 2020-04-07:
public void startConn(BleDevice bleDevice) {

        BleManager.getInstance().connect(bleDevice, new BleGattCallback() {
            @Override
            public void onStartConnect() {
                Ui.tip(context, context.getString(R.string.CONN_START));
//                Log.i("tag", "开始链接");
            }

            @Override
            public void onConnectFail(BleDevice bleDevice, BleException exception) {
                Ui.tip(context, context.getString(R.string.CONN_FAILE));
//                Log.i("tag", "链接失败");
            }

            @Override
            public void onConnectSuccess(BleDevice bleDevice, BluetoothGatt gatt, int status) {
//                Log.i("tag", "链接成功|" + bleDevice.getMac());
                Ui.tip(context, context.getString(R.string.CONN_SUCCESS));
                //存储链接成功的mac
                context.saveMac(bleDevice.getMac());
                App.deviceMac = bleDevice.getMac();
                App.deivce = bleDevice;
                if (!mDevice.contains(bleDevice)) mDevice.add(bleDevice);
//                for (BluetoothGattService service : gatt.getServices()) {
//                    Log.i("tag", "service|" + service.getUuid());
//                    for (BluetoothGattCharacteristic chara :
////                            gatt.getService(UUID.fromString(context.getInfo().cmut_uuid)).getCharacteristics()
//                            service.getCharacteristics()
//                    ) {
//                        Log.i("tag", "oo|" + chara.getUuid());
//                    }
//                }
                //6e400001-b5a3-f393-e0a9-e50e24dcca9e

                try {

                    String uuid = context.getInfo().cmut_uuid;
                    String uuid_w = context.getInfo().cmut_uuid_w;
                    String uuid_n = context.getInfo().cmut_uuid_r;
//                    String uuid = BLE.LG_UUID_SERVICE;
//                    String uuid_w = BLE.LG_UUID_WRITE;
//                    String uuid_n = BLE.LG_UUID_NOTIFY;

                    App.UUID_SERVICE = gatt.getService(UUID.fromString(uuid));
                    UUID_SERVICE = App.UUID_SERVICE;

                    App.UUID_WRITE = App.UUID_SERVICE.getCharacteristic(UUID.fromString(uuid_w));
                    UUID_WRITE = App.UUID_WRITE;

                    App.UUID_NOTIFY = App.UUID_SERVICE.getCharacteristic(UUID.fromString(uuid_n));
                    UUID_NOTIFY = App.UUID_NOTIFY;
                } catch (Exception e) {
                    Ui.tip(context, context.getString(R.string.BluetoothUnable1));
                    App.GATT = null;
                    mGatt = null;
                    gatt.disconnect();
                    gatt.close();
                    return;
                }

                if (App.UUID_WRITE == null) {
                    gatt.disconnect();
                    gatt.close();
                    return;
                }
                boolean connState = gatt.setCharacteristicNotification(App.UUID_NOTIFY, true);

//                //频率ok
                if (connState) {
[... 64 lines cut ...]
        for (byte element : bytes) {
            int v = element & 0xff;
            if (v < 16) {
                sb.append('0');
            }
            sb.append(Integer.toHexString(v));
        }
        return sb.toString().toUpperCase(Locale.US);
    }

    void onBleConnSetNotifyOff(BleDevice bleDevice, String serviceId, String charId) {
        BleManager.getInstance().stopNotify(bleDevice, serviceId, charId);
    }

    public void sendPacket(byte[] data) {
        try {
            App.UUID_WRITE.setWriteType(BluetoothGattCharacteristic.WRITE_TYPE_NO_RESPONSE);
            boolean result = App.UUID_WRITE.setValue(data);
            boolean res = App.GATT.writeCharacteristic(App.UUID_WRITE);
//        Log.i("tag", "result|" + res);
//            Log.i("tag", "result||" + res);
        } catch (Exception e) {

        }
    }

kingiis, 2020-04-07:
蓝牙数据发送时可以根据boolean值得知发送成功失败,但是得到返回是回调函数,如果你想同步处理就要先主线程沉睡,因为返回值是异步的
When sending Bluetooth data, you can know the success or failure of sending according to the Boolean value, but the return function is the callback function. If you want to synchronize, you need to sleep the main thread first, because the return value is asynchronous

kingiis, 2020-04-07:
or You are not communicating with SSP socket!