Jasonchenlijian/FastBle #481: 华为手机 安卓10.1版本，不能扫描身边蓝牙

duanxin1990, 2020-06-04:
华为手机安卓版本升级到最新的10.1，安装自己的工程，发现蓝牙扫描不到身边的蓝牙设备，之前安卓10版本是可以扫描到的

yingzikeji, 2020-06-15:
[image]
[image]

` add permission   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>`

lxfwelldone, 2020-08-31:
第一次集成， 华为 10.1也遇到这个问题了， 请问怎么解决的，我的刚好是28，确定已经申请了定位和蓝牙权限

yingzikeji, 2020-08-31:
安卓10需要开启定位权限，你把定位权限申请，并打开权限就OK了。-------- 原始邮件 --------发件人： Richard <notifications@github.com>日期： 2020年8月31日周一 下午3:36收件人： Jasonchenlijian/FastBle <FastBle@noreply.github.com>抄送： Enoch <yingzikeji@gmail.com>, Comment <comment@noreply.github.com>主    题： Re: [Jasonchenlijian/FastBle] 华为手机 安卓10.1版本，不能扫描身边蓝牙 (#481)
第一次集成， 华为 10.1也遇到这个问题了， 请问怎么解决的，我的刚好是28，确定已经申请了定位和蓝牙权限

lxfwelldone, 2020-08-31:
定位打开i的，已经申请了权限，不知道是不是Android X的原因，我在另外一个非X项目中可以扫描的；
但是指定uuid就扫描不到，和您的项目中的apk一样指定uuid就扫描不到，好像不止我一个人遇到

lxfwelldone, 2020-09-01:
我把项目清理了一遍，可以了

rock0201, 2020-12-09:
.ACCESS_FINE_LOCATION 打开这个权限就可以运行 亲测