Jasonchenlijian/FastBle #487: Notify / Indicate doubled twice

hassance3, 2020-07-20:
When I sent a "Notify" from my BLE device, the FastBLE app create two events, `BleManager.getInstance().notify(...)` wish is okay, and `BleManager.getInstance().indicate()` wish is a mistake.

whatever a Notify or Indicate is sent, FastBLE app create two events every time.

kingiis, 2020-11-16:
我通常只选择 notify的方式来处理，全频道广播。
I usually only choose the way of notify to deal with all channel broadcasting

kingiis, 2020-11-16:
public class MainApplication extends Application {
...
public static BleDevice deivce;
    public static BluetoothGatt GATT;
    public static BluetoothGattService UUID_SERVICE;
    public static BluetoothGattCharacteristic UUID_WRITE;
    public static BluetoothGattCharacteristic UUID_NOTIFY;
...
}
1.BleManager.getInstance().connect(MainApplication.deivce, object : BleGattCallback() {
...
       override fun onConnectSuccess(bleDevice: BleDevice?, gatt: BluetoothGatt?, status: Int) {
                MainApplication.deviceMac = bleDevice!!.mac
                MainApplication.deivce = bleDevice
                MainApplication.GATT = gatt
...
}
2. val connState: Boolean = MainApplication.GATT.setCharacteristicNotification(MainApplication.UUID_NOTIFY, true)
        if (connState) {
            BleManager.getInstance().notify(MainApplication.deivce,
                    MainApplication.UUID_SERVICE.uuid.toString(),
                    MainApplication.UUID_NOTIFY.uuid.toString(),
                    object : BleNotifyCallback() {
...
 override fun onNotifySuccess() {
                            Logi.i("订阅了成功了")
...
}
}

I use the ingenious scheme, first test whether the connection can be made, and then save the device object, directly connect to the subscription