Jasonchenlijian/FastBle #493: Ble devices are getting disconnected within 3-4 sec after connection.

pritam2102, 2020-11-03:
Hi,

We are  facing some issues between our ble devices (ble version 5) and android mobiles using FastBle library.

We are able to connect our devices with mobile phones like samsung, redmi note 8 pro, redmi not 5 pro and other mobiles.

But when we are trying to connect same devices with mobiles like redmi note 8, redmi note 9 pro max and oppo A53. **it is getting disconnected within 3-4 sec with gatt timeout error 8.**

After tracing bluetooth snoop log we can see below error.
Attribute not found 
[image]
 {
13274	12783.807351	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
13275	12783.808066	remote ()	XiaomiCo_de:0a:37 (Redmi note 9 pro max)	L2CAP	497	Rcvd Connection oriented channel
13276	12783.837010	Microchi_95:fc:5a (Endocare)	XiaomiCo_de:0a:37 (Redmi note 9 pro max)	ATT	74	Rcvd Read By Type Response, Attribute List Length: 3, Unknown, Unknown, Unknown
13277	12783.839225	XiaomiCo_de:0a:37 (Redmi note 9 pro max)	Microchi_95:fc:5a (Endocare)	ATT	16	Sent Read By Type Request, GATT Characteristic Declaration, Handles: 0x0057..0x0058
13278	12783.842390	controller	host	HCI_EVT	8	Rcvd Number of Completed Packets
13279	12783.858398	Microchi_95:fc:5a (Endocare)	XiaomiCo_de:0a:37 (Redmi note 9 pro max)	ATT	14	Rcvd Error Response - Attribute Not Found, Handle: 0x0057 (Unknown: Unknown)
}

We have already tried changing connection parameters like interval, latency and timeout but no luck.

We have attached screen shot of  hci snoop log file.

Please suggest on this.

kingiis, 2020-11-16:
我把链接用的东西都放在application里
当onDestroy()不用蓝牙时

   BleManager.getInstance().disconnect(MainApplication.deivce) 
            MainApplication.GATT.disconnect()
            MainApplication.GATT = null
            MainApplication.UUID_SERVICE = null
            MainApplication.UUID_WRITE = null
            MainApplication.UUID_NOTIFY = null

希望对你有帮助

pritam2102, 2020-11-28:
Thank You,
I tried but not working.

Jasonchenlijian, 2020-12-23:
It's about your device. Some devices will automatically disconnect within a few seconds if they do not communicate after a successful connection.