Jasonchenlijian/FastBle #494: bt_btm_sec: btm_sec_disconnected clearing pending flag handle:512 reason:8 自动断开连接了

Mr-Smile, 2020-12-15:
bt_stack: [INFO:btsnoop.cc(323)] clear_l2cap_whitelist: Clearing whitelist from l2cap channel. conn_handle=512 cid=4:4
11-26 19:06:02.139589  3058  3663 W bt_stack: [WARNING:bta_gattc_act.cc(1045)] bta_gattc_conn_cback: cif=3 connected=0 conn_id=0x0003 reason=0x0008
11-26 19:06:02.139705  3058  3663 W bt_stack: [WARNING:bta_gattc_act.cc(1045)] bta_gattc_conn_cback: cif=4 connected=0 conn_id=0x0004 reason=0x0008
11-26 19:06:02.139780  3058  3663 W bt_stack: [WARNING:bta_gattc_act.cc(1045)] bta_gattc_conn_cback: cif=6 connected=0 conn_id=0x0006 reason=0x0008
11-26 19:06:02.139909  3058  3663 I bt_btm_sec: btm_sec_disconnected clearing pending flag handle:512 reason:8
11-26 19:06:02.140574   502   502 D [BT]    : mtk_bt_op: BT_VND_OP_LPM_WAKE_SET_STATE
11-26 19:06:02.140660  3058  3192 I btif_av : btif_av_acl_disconnected: Peer 86:3a:c9:d5:ca:37 : ACL Disconnected
11-26 19:06:02.140916  3058  3663 E btif_av : FindOrCreatePeer: acquire the lock.
11-26 19:06:02.140973  3058  3663 I btif_av : BtifAvPeer *BtifAvSource::FindOrCreatePeer(const RawAddress &, tBTA_AV_HNDL): Create peer: peer_address=86:3a:c9:d5:ca:37 bta_handle=0x41 peer_id=0
11-26 19:06:02.141043  3058  3663 W bt_btif : btif_av_get_peer_sep: No active peer found
11-26 19:06:02.141077  3058  3663 I bt_btif_a2dp: btif_a2dp_on_idle: ## ON A2DP IDLE ## peer_sep = 1
11-26 19:06:02.141103  3058  3663 W bt_btif : btif_av_get_peer_sep: No active peer found
11-26 19:06:02.141121  7975  7992 D BluetoothGatt: onClientConnectionState() - status=8 clientIf=6 device=86:3A:C9:D5:CA:37
11-26 19:06:02.141138  3058  3663 I bt_btif_a2dp_source: btif_a2dp_source_on_idle: state=STATE_OFF
11-26 19:06:02.141196  3058  3663 W bt_btif : btif_av_get_peer_sep: No active peer found
11-26 19:06:02.141223  3058  3663 I bt_btif_a2dp: btif_a2dp_on_idle: ## ON A2DP IDLE ## peer_sep = 1
11-26 19:06:02.141250  3058  3663 W bt_btif : btif_av_get_peer_sep: No active peer found
11-26 19:06:02.141275  3058  3663 I bt_btif_a2dp_source: btif_a2dp_source_on_idle: state=STATE_OFF
11-26 19:06:02.141329  3058  3663 I btif_av : DeleteIdlePeers: Deleting idle peer: 86:3a:c9:d5:ca:37 bta_handle=0x41
11-26 19:06:02.146398  7975  7975 D BluetoothGatt: cancelOpen() - device: 86:3A:C9:D5:CA:37
11-26 19:06:02.148600  7975  7975 D BluetoothGatt: refresh() - device: 86:3A:C9:D5:CA:37
11-26 19:06:02.148691  3081  3159 W BluetoothEventManager: AclStateChangedHandler: activeDevice is null
11-26 19:06:02.149159  4665  4665 W BluetoothEventManager: AclStateChangedHandler: activeDevice is null
11-26 19:06:02.149597  3058  3663 E bt_stack: [ERROR:bta_gattc_act.cc(294)] No such connection need to be cancelled
11-26 19:06:02.149788  3058  3663 I bt_stack: [INFO:gatt_api.cc(1183)] GATT_CancelConnect: gatt_if:6, address: 86:3a:c9:d5:ca:37, direct:0
11-26 19:06:02.150455  7975  7975 D BluetoothGatt: close()
11-26 19:06:02.150572  7975  7975 D BluetoothGatt: unregisterApp() - mClientIf=6
11-26 19:06:02.152141  3058  3663 I bt_stack: [INFO:bta_gattc_act.cc(215)] bta_gattc_deregister: whitelistsize 64 39
11-26 19:06:02.152387   784   854 D AAL     : 11-26 07:06:01.963 BL= 323,ESS= 256, 11-26 07:06:01.995 BL= 324,ESS= 256, 11-26 07:06:02.041 BL= 325,ESS= 256, 11-26 07:06:02.072 BL= 326,ESS= 256, 11-26 07:06:02.152 BL= 327,ESS= 256, 
11-26 19:06:02.152376  7975  7975 D HZH     : onDisConnected: 

===================================================================================

正连接的时候莫名的断开连接，android 10、android 11都出现过

mikaelzero, 2022-03-22:
请问解决了吗

emily800326, 2022-06-14:
也出現No active sink or source peer found 同問題
想詢問怎麼解決