Jasonchenlijian/FastBle #501: 请教下应该如何提高蓝牙传输速率？

Shirley0207, 2021-02-24:
我这边要实现发送文件包，目前已实现，不过速率大概是1kb/s，感觉还是太慢了，一个40KB的文件要发40秒。
请问有什么方法可以提高传输速率吗？
目前实现的方法中没有设置mtu，默认20个字节一个包。
有一个疑问就是扩大mtu会提高这个传输速率吗？但是我尝试了设置mtu，但也不起作用，还是20个字节一个包。设置mtu是不是也要看蓝牙设备的？

Shirley0207, 2021-02-27:
设置了MTU，onMtuChanged回调还是23，是说明这个设备不接受扩展是嘛，但是我用其它蓝牙调试助手发数据，设备还是能一次性接收超过20字节的数据。
设置过splitWriteNum，发送数据时，回调会显示写入数据成功，但是实际上设备只接收到前20个字节，其它数据都丢失了
有没有大神可以帮忙一下

hmxbanz, 2021-03-07:
打开蓝牙高速模式

hmxbanz, 2021-03-10:
不正常，硬件可以打开高速。安卓也可以打开高速。任意一方操作都可以。MTU只是包大小规格，和速度无关。requestConnectionPriority方法

kingiis, 2021-03-31:
看硬件级别，一般 蓝牙4.1 4.2 5.0 MTU到512Byte
我测过的速度大概是每分钟1MB
如果对速度有要求，板子直接wifi热点和蓝牙一起开，数据走wifi，命令走蓝牙

zhubinsheng, 2021-08-23:
> > 打开蓝牙高速模式
> 
> 这个指的是硬件打开高速模式吗？还是说App可以打开蓝牙设备的高速模式啊？
> 
> 还想问一下，我和硬件沟通过了，他修改了硬件，现在可以成功修改mtu了。我把mtu设置为247，splitWriteNum设置为244，这样的话传输速率大概在13kb/s，请问这样的速率是正常的吗？

正常的，低功耗蓝牙透传速度差不多