Jasonchenlijian/FastBle #506: Split on false with data that is larger than 512 max mtu

ymebrugts, 2021-03-11:
Hi,

What happens if I set split to `false` and the data is larger than 512 bytes? 

From the doc:
`Without expanding MTU and expanding MTU's ineffectiveness, subcontracting is required when long data with more than 20 bytes are to be sent. The parameter boolean split indicates whether to use packet delivery; the write method without the boolean split parameter is subcontracted to the data by more than 20 bytes by default.`

Greetings,
Yme

kingiis, 2021-03-31:
板子上边主动分包啊,兄弟
var  x=  MTU(max value)
板子根据包体最大主动信息分段,比如最大是512byte.板子就要在480byte分段，头尾分别按照你们自己的协议添加分段信息，例如，下标和总共多少条

chen-fuhai, 2021-08-24:
OtherException: requiredMtu should lower than 512 !