Jasonchenlijian/FastBle #510: FastBle 写入异常Code= 102，如何解决？

wzy888, 2021-07-29:
onWriteFailure: BleException { code=102, description='gatt writeCharacteristic fail'}

wangyuan0217, 2021-08-14:
同问啊，用默认的MTU就不回出现这个问题，修改了 MTU为最大就不行了

chen-fuhai, 2021-08-24:
重连

kingiis, 2021-08-24:
MTU没毛病的 ，要在订阅读写后，去申请完毕再 读写数据

wangyuan0217, 2021-08-24:
> onWriteFailure: BleException { code=102, description='gatt writeCharacteristic fail'}

一般在连接成功不能立马setMtu  或者立马 write 。 ble连接成功建立通信后要加个短暂的delay在setMtu或者write，不然会102。

330132662, 2021-09-09:
插眼

massimilianochiodi, 2021-09-23:
A pause after detecting the device and connecting solves problem 102

kingiis, 2021-09-30:
> 
> 
> onWriteFailure: BleException { code=102, description='gatt writeCharacteristic fail'}

哦 对的  所有蓝牙通病 100ms延迟

twilight2dev, 2024-07-31:
anything new? i'm still having this issue, even with the delay added

wang6793453, 2024-08-07:
一定要先订阅再去写入

hcyAndroid, 2025-02-19:
我觉得跟什么先订阅再写入没啥关系，进瞎扯淡

[image]

hcyAndroid, 2025-02-19:
如果你频繁的发送数据，发送数据要有一定的间隔。不然就有这个问题。这个库从处理到回调有一定时间，蓝牙硬件ble固件也有数据间隔要求。

hcyAndroid, 2025-02-19:
当然不排除与硬件约定的原因。举个例子，硬件那边需要的数据要根据mtu进行分包再发过去

kingiis, 2025-02-21:
对待硬件的所有操作要有延迟，最好做一个队列任务,或者 子线程操作+delay 100ms-200ms。
这种通用协议即使是 菊花厂都没那闲情逸致改这玩意封装。

jinle, 2025-06-27:
在我的手机上，修改为如下代码后，不再出现102错误：
BleManager.getInstance().write(mBleDevice, SERVICE_ID, CHARACT_ID, bytes, false, new BleWriteCallback() { ... });