Jasonchenlijian/FastBle #512: Vivo S1 android 11 issue

somanisushant23, 2021-08-19:
This does not work for Vivo S1 phones.Can you dig a little deeper in this?

massimilianochiodi, 2021-09-23:
Your Vivo have 'Android one' SO ? 
Localization is enable for app ? bluetooth scan not work if disabled 
In permission use (Android MANIFEST) : 

<uses-permission
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN" />
    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/> ***** ANDROID 12 ***** 
    <uses-permission android:name="com.google.android.things.permission.MANAGE_BLUETOOTH" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/> 
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    
**Bluetooth Connectivity Problems**
Open the Bluetooth settings on your phone. Check if something has been customized. The same can be a reason to this glitch
Clear the Bluetooth history from your phone and check if the Bluetooth start working properly after doing so
Reboot your phone and check if the problem is gone
There are chances that the problem is there just because the phone communication is disabled by you manually. Remove all other restrictions as well
Make sure the device you are going to connect with is not having something wrong with it.
A virus infection can be a reason you are unable to connect your device with another through Bluetooth
The reason you are facing this problem might be related with the phone OS. Check it and fix the problem you locate
There are chances that the phone Bluetooth antenna is not in working condition
Sometimes the users have to face this problem just because the power saving or the safe mode is turned ON
This problem can declare its presence when you don’t clean the cache data from your phone
There are chances that the storage space is lacking in your device and is causing this problem. Check the other device as well
There are certain mistakes that users make while pairing the devices with each other for the very first time. One example is entering the wrong pass code. Check if this is the case with you as well
Make sure the actual problem is not related with the Bluetooth compatibility
There are chances that the data which you are using is corrupt or is not supported by any or both the devices
Check if a simple restart solves this problem easily from your phone

kingiis, 2021-09-30:
你是搜索不到蓝牙还是搜索不到特定设备？
先检查权限，然后添加对照组手机，华为，小米，板子准备多个。
不排除板子代码问题导致的搜索失败。我遇到过的