Jasonchenlijian/FastBle #515: 为什么魅族5.1扫描不到指定的mac，重启APP也扫描不到，一直重复mAddress，6.0以上就没问题

chen-fuhai, 2021-09-01:
I/MainActivity: onResume: 4D:57:4B:5A:59:42
D/BluetoothAdapter: isEnabled
D/BluetoothAdapter: startLeScan(): null
D/BluetoothAdapter: 748199934: getState(). Returning 12
D/BluetoothAdapter: 748199934: getState(). Returning 12
D/BluetoothAdapter: 748199934: getState(). Returning 12
D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=4
I/ApiConfig: onScanStarted: true
I/MainActivity: onScanStarted: 
D/BluetoothDevice: mAddress: 41:47:53:4A:53:58
D/BluetoothDevice: mAddress: 41:47:53:4A:53:58
I/ApiConfig: onLeScan: 41:47:53:4A:53:58
D/BluetoothDevice: mAddress: 41:47:53:4A:53:58
D/BluetoothDevice: mAddress: 41:47:53:4A:53:58
I/ApiConfig: onLeScan: 41:47:53:4A:53:58
D/BluetoothDevice: mAddress: 41:47:53:4A:53:58
D/BluetoothDevice: mAddress: 41:47:53:4A:53:58
I/ApiConfig: onLeScan: 41:47:53:4A:53:58

kingiis, 2021-09-30:
> I/MainActivity: onResume: 4D:57:4B:5A:59:42 D/BluetoothAdapter: isEnabled D/BluetoothAdapter: startLeScan(): null D/BluetoothAdapter: 748199934: getState(). Returning 12 D/BluetoothAdapter: [...]

检查 定位权限和蓝牙权限 多自测，蓝牙这个主从机模式,很难说谁一定是对的.
多添加几个对照组设备,我遇到过 华为手机可以搜索到，小米就搜索不到，后来板子改代码就好了

kingiis, 2021-09-30:
你可以默认搜索,找到就用 断开就走回调断开。
当一个设备太频繁的时候 Android这边会把你踢掉。板子那边会出现卡死的情况

chen-fuhai, 2022-01-05:
> I/MainActivity: onResume: 4D:57:4B:5A:59:42 D/BluetoothAdapter: isEnabled D/BluetoothAdapter: startLeScan(): null D/BluetoothAdapter: 748199934: getState()。返回 12 D/BluetoothAdapter：748199934：getState()。返回 12 D/BluetoothAdapter：748199934：getState()。返回 12 D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=4 I/ApiConfig: onScanStarted: true I/MainActivity: onScanStarted: D/BluetoothDevice: mAddress: 41:47:53:4A:53:58 D/BluetoothDevice: mAddress: 41:47:53:4A:53:58 I/ApiConfig: onLeScan: 41:47:53:4A:53:58 D/BluetoothDevice: mAddress: 41:47:53:4A:53:58 D/BluetoothDevice : m 地址: 41:47:53:4A:53:58 I/ApiConfig: onLeScan: 41:47:53:4A:53:58 D/BluetoothDevice: mAddress: 41:47:53:4A:53:58 D/BluetoothDevice: mAddress: 41:47:53:4A:53: 58 I/ApiConfig: onLeScan: 41:47:53:4A:53:58

closed 2022-01-05 by chen-fuhai