Jasonchenlijian/FastBle #516: hexa

Hashim-svg, 2021-09-01:
The project is awesome but the issue is you are sending and receiving hex values but how can we send simple text. the esp32 doesn't have enough space that we implement hex to text inside it if we implement it there will be not enough  space to implement other things .....

massimilianochiodi, 2021-09-23:
I don't know if I understand correctly ...

Let's see if I understand.
You have to send some text ...
So you need to convert the text to hex to send it

But you say you don't have space?
In what sense ? the string you have to send doesn't fit in 20 bytes?
Because you can increase MTU and send your string anyway. 
Then, if on the other side ... the receiver uses a NOTIFICATION system for receive, you should send the text separated by groups of 250 bytes or maybe more ( depend on MTU )
MTU == Max Trasmission Unit ( normally in bytes )

kingiis, 2021-09-30:
兄弟 你是不是没明白这个ble得功能流程
你通过api 无论发送什么数据,下边都是ascii在传递,二进制字节码.
你什么空间不够？传递一次十几,二十几个字节,我还嫌少呢.你硬件支持上来了 每秒能传1-2Mbps都可以。

kingiis, 2021-09-30:
> 
> 
> 该项目很棒，但问题是您正在发送和接收十六进制值，但我们如何发送简单的文本。 esp32 没有足够的空间来实现十六进制到其中的文本，如果我们实现它，将没有足够的空间来实现其他东西
> 
> The project is awesome but the issue is you are sending and receiving hex values but how can we send simple text. the esp32 doesn't have enough space that we implement hex to text inside it if we implement it there will be not enough space to implement other things .....

在ble这个套路中,我一般都用 16进制字符串 得形式编辑.
你看,默认入参是  byte[],如果 tostring后 你能看到得是  ascii的数字,可读性很低的。
16进制字符串方便编辑,比如 _ _ ，00-FF，肉眼就能知道是什么东西 可读性很强,编辑性更强,后台存储以 ： 符号分割。
至于工具类  16进制字符串转byte[] 的方法很多的,中文补位都支持的。