Jasonchenlijian/FastBle #518: 如何根据多个mac地址扫描

BaoBaoss, 2021-09-21:
如何根据多个mac地址扫描蓝牙设备，貌似只能配置一个mac

massimilianochiodi, 2021-09-22:
You can scan by applying the filter for a service or characteristic

Example for java :  
`
public static final UUID BLUETOOTH_WALLBOX_SERVICE = UUID.fromString( "331a36f5-2459-45ea-9d95-6142f0c4b307");

UUID[] listauuid = new UUID[] {Costanti.BLUETOOTH_WALLBOX_SERVICE};
    String device_name = "";
    String device_mac = "";
    BleScanRuleConfig scanRuleConfig = new BleScanRuleConfig.Builder()
            .setServiceUuids( listauuid )
            .setDeviceName(true, device_name )
            .setDeviceMac( device_mac )
            .setAutoConnect(false)
            .setScanTimeOut(BLUETOOTH_WALLBOX_SCANPERIOD)
            .build();
    BleManager.getInstance().initScanRule(scanRuleConfig);

    BleManager.getInstance().scan(new BleScanCallback() {
.....
`

massimilianochiodi, 2021-09-22:
BLUETOOTH_WALLBOX_SCANPERIOD = 5000; !

BaoBaoss, 2021-09-22:
> You can scan by applying the filter for a service or characteristic
>
> Example for java :
>
> ```
> public static final UUID BLUETOOTH_WALLBOX_SERVICE = UUID.fromString( [...]

不知道为什么，我的蓝牙设备不能直接得到UUID，只能通过解析ScanRecord得到，通过UUID过滤是不成功的

massimilianochiodi, 2021-09-23:
Your device probably has a dynamic UUID, that is, it can change or it has hidden uuid Or it is not yet a 'commercial' device or correctly registered with the bluetooth consortium... 

It is not a problem that cannot be solved:
1) you must know the advertising record perfectly
2) you must always read all nearby devices during the scan, because you have to compare all scan records (adverstising record)
**Then surely only by knowing your enemy can you win the battle!**

Can you tell me which is your device?
Maybe, I can help you, you see ... I wrote the app for the wallbox of one of the most important car manufacturers in the world that uses a custom bluetooth.

BaoBaoss, 2021-09-25:
> Your device probably has a dynamic UUID, that is, it can change or it has hidden uuid Or it is not yet a 'commercial' device or correctly registered with the bluetooth consortium...
>
> It is not a [...]

我使用安卓原生的api进行扫描发现身边所有的蓝牙设备获得的name和uuis全都是null，只有address有值，如图所示：
其中scanFilters我没有添加任何项目，buildDefaultScanSettings函数使用默认参数
[image]
[image]
日志输出如下：
[image]
我参考了厂家给出的实例apk，按照这个方法解析ScanRecord类获得的字节码就能得到UUID，其中scanData数组是ScanRecord.getBytes()方法获得的值
[image]
所以根据uuid来扫描设备永远扫描不到任何设备，只有根据address才可以

kingiis, 2021-09-30:
> 
> 
> 如何根据多个mac地址扫描蓝牙设备，貌似只能配置一个mac

你可以默认搜索,找到就用 断开就走回调断开。
当一个设备太频繁的时候 Android这边会把你踢掉。板子那边会出现卡死的情况