Jasonchenlijian/FastBle #538: Bugly监测到一个空指针异常

932707629, 2022-04-02:
版本：2.4.0

java.lang.RuntimeException:Unable to stop activity {com.lianxin.oyt/com.lianxin.oyt.ui.GpBleDeviceListActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.os.HandlerThread.quit()' on a null object reference
android.app.ActivityThread.callActivityOnStop(ActivityThread.java:5224)
......
Caused by:
java.lang.NullPointerException:Attempt to invoke virtual method 'boolean android.os.HandlerThread.quit()' on a null object reference
com.clj.fastble.scan.BleScanPresenter.notifyScanStopped(BleScanPresenter.java:207)
com.clj.fastble.scan.BleScanner.stopLeScan(BleScanner.java:131)
com.clj.fastble.BleManager.cancelScan(BleManager.java:385)
com.lianxin.oyt.ui.GpBleDeviceListActivity.onStop(GpBleDeviceListActivity.java:167)
android.app.Instrumentation.callActivityOnStop(Instrumentation.java:1526)
android.app.Activity.performStop(Activity.java:8344)
android.app.ActivityThread.callActivityOnStop(ActivityThread.java:5216)
android.app.ActivityThread.performStopActivityInner(ActivityThread.java:5195)
android.app.ActivityThread.handleStopActivity(ActivityThread.java:5273)
android.app.servertransaction.TransactionExecutor.performLifecycleSequence(TransactionExecutor.java:233)
android.app.servertransaction.TransactionExecutor.cycleToPath(TransactionExecutor.java:201)
android.app.servertransaction.TransactionExecutor.executeLifecycleState(TransactionExecutor.java:173)
android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:97)
android.app.ActivityThread$H.handleMessage(ActivityThread.java:2323)
android.os.Handler.dispatchMessage(Handler.java:106)
android.os.Looper.loop(Looper.java:257)
android.app.ActivityThread.main(ActivityThread.java:8243)
java.lang.reflect.Method.invoke(Native Method)
com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:626)
com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1031)

tinlen, 2022-04-29:
解决了吗，我也监控到了

LimYuChan, 2022-09-29:
if(BleManager.getInstance().scanSate == BleScanState.STATE_SCANNING){
      BleManager.getInstance().cancelScan()
    }