Jasonchenlijian/FastBle #559: 关闭手机蓝牙，再次开启蓝牙后，重新连接成功后回调中bleDevice对象下的BluetoothDevice 对象为空

zhicong666, 2022-09-30:
1、搜索蓝牙后得到的蓝牙对象，在关闭手机蓝牙后，所有已搜索的蓝牙bleDevice下的BluetoothDevice 会变成空
2、关闭手机蓝牙，再次开启蓝牙后，重新连接成功后回调中bleDevice对象下的BluetoothDevice 对象为空
3、正常远距离断开重连是正常的

希望方便的时候帮忙分析一下原因

 ///<summary>
    ///蓝牙连接
    ///</summary>
    ///<param name="bleDevice">蓝牙设备</param>
    fun connect(bleDevice: BleDevice) {
        BleManager.getInstance().connect(bleDevice, object : BleGattCallback() {
            override fun onStartConnect() {
                println("---------------> onStartConnect")
            }

            override fun onConnectFail(bleDevice: BleDevice, exception: BleException) {

            }

            override fun onConnectSuccess(
                bleDevice: BleDevice,
                gatt: BluetoothGatt,
                status: Int
            ) {
                // bleDevice.device 对象为空
            }

            override fun onDisConnected(
                isActiveDisConnected: Boolean,
                bleDevice: BleDevice,
                gatt: BluetoothGatt,
                status: Int
            ) {

            }
        })
    }

Wheats, 2022-10-17:
这个bug同样遇到，

解决方法：

1:在打开蓝牙的时候，要么等待蓝牙自动搜索附近设备 这个时间一般在(500ms-1200ms)可以完成。看手机系统

2:打开蓝牙后，手动调用搜索接口，重新装载蓝牙

问题出在，扫描设备到mac地址，到时mac映射成BluetoothDevice信息，mScan。需要时间，

测试过，关闭后直接连接，有mac信息，但是name为null。等待1-2秒装载后，在连接，mac 跟 name 就有值了，

new-soul, 2023-11-13:
还是重新scan再连接比较保险，这样是可以正常。

Wheats, 2023-11-19:
Android 14 测试发现，断开蓝牙时，name为null 
测试手机 google pixel 6 pro
系统 Android 14

liuyuanhan, 2023-12-20:
关闭蓝牙，并重新打开后，会有部分Android机型出现，旧的已经连接的BleDevice.getName()为null，但实际存储在MultipleBluetoothController哈希表中的BleBluetooth依然没变
因此，当我们重新扫描蓝牙，扫描到新的设备进行连接时，实际上并没有更新在MultipleBluetoothController的哈希表中的BleBluetooth, 依然是使用断开蓝牙前的BleBluetooth, 所以会出现name为null，但是连接又成功的情况
目前我的解决办法是，将BleDevice中的getKey()方法进行修改，去掉名字前缀，以mac地址为准，即可解决

liuyuanhan, 2023-12-20:
[image]

liuyuanhan, 2023-12-20:
[image]

Wheats, 2023-12-21:
一些，系统在打开蓝牙时刻，获取蓝牙设备名称会获取NULL，针对这个问题，我监听了蓝牙开关，在打开的时候，重新调用扫描方法，这个问题得到了90%解决吧

 
1551703036554_.pic.jpg (view on web)

liuyuanhan, 2023-12-21:
重新扫描后，会出现一个新的问题，经过验证，在三星和google的手机上，当关闭蓝牙时，对于已经连接过的设备，removeBleBluetooth()方法并不会调用，但是国产机基本都会调用。这就会导致扫描到设备进行重新连接时，会出现连接成功，但是setMtu和发现服务会失败的情况。
所以一旦手动关闭蓝牙，需要在蓝牙关闭的通知里面，将BleBluetooth移除掉。