Jasonchenlijian/FastBle #574: 我已解决notify的那个102 descriptor equals null 的错

wangjinchan, 2023-06-16:
调试了一会儿，发现作者使用的mCharacteristic是空的，而不是从函数传入的。所以我就在BleManager.getInstance().notify这个函数里加上了characteristic；详情请看我改了哪些地方，跟着我改就可以了：
[image]
[image]
[image]
[image]
[image]
[image]
` BluetoothGattDescriptor descriptor = null;
        List<BluetoothGattDescriptor> descriptors = characteristic.getDescriptors();
        for (BluetoothGattDescriptor desc : descriptors) {
            if (desc.getCharacteristic().getUuid().equals(formUUID(UUID_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR))) {
                descriptor = desc;
                break;
            }
        }`

wangjinchan, 2023-06-16:
如果还有什么没看懂 ，可以问 我，邮箱：945750315@qq.com
希望作者把我拉进去一起维护。

wwxwwq, 2023-06-28:
[image]
这里mCharacteristic不是已经赋值了吗

wangjinchan, 2023-06-28:
没有，那个值是空的

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Wenxiang Wang ***@***.***&gt;
发送时间: 2023年6月28日 18:54
收件人: Jasonchenlijian/FastBle ***@***.***&gt;
抄送: wangjinchan ***@***.***&gt;, Author ***@***.***&gt;
主题: Re: [Jasonchenlijian/FastBle] 我已解决notify的那个102 descriptor equals null 的错  (Issue #574)

wwxwwq, 2023-06-29:
连接上后，扫描到服务，再开启通过，这里应该不可能空
[image]

wangjinchan, 2023-06-29:
就是开始那个读的服务时报错了，我一层一层去打点调试出来的，改了之后，才能正常读取外围设备发的数据。

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: Wenxiang Wang ***@***.***&gt;
发送时间: 2023年6月29日 13:54
收件人: Jasonchenlijian/FastBle ***@***.***&gt;
抄送: wangjinchan ***@***.***&gt;, Author ***@***.***&gt;
主题: Re: [Jasonchenlijian/FastBle] 我已解决notify的那个102 descriptor equals null 的错  (Issue #574)

hiddus, 2023-09-06:
每次都是newBleConnector characteristic每次都会赋值。
我这边状况是，有时正常，读写发送都好着。

alvayang, 2023-10-02:
非常感谢 @wangjinchan 通过调查发现，很可能是下位机对于一条指令的结束没有给出正确的反应，比价合适的方式是修改MTU的大小，当然下位机是需要能支持的。
通过和 @wangjinchan 的讨论，建议收不到notify的时候，改改MTU试试看效果。

ThanatosDan, 2023-10-10:
[image]
liangg两个一样的UUID 第二个才是notify 导致了这个问题

ThanatosDan, 2023-10-10:
[image]
[image]
我这边这样改好像也可以